coassoftwaresystems/delphi-modbus #3: Error 10004

alvaropag, 2015-11-23:
Hello,

I'm using the last version (1.6.5) with delphi XE8 + Indy 10 and I don't know why but the communication stops working (in a read request). I wasn't able to pin point exactly the problem, but when I stop the debugger a "Socket error 10004 Interrupted system call" is raised by the TIdIOHandler.ReadFromSource function. And althought I set a Read Timeout of 4s it doesn't seem to be working (even though the timeout parameter is correctly set inside the function).
I'd like to know if someone already experienced this and how could I handle it.
Thanks.

P.S.: tested on windows 7 - x64.

gcser, 2016-05-04:
Hello,

I just experiencing the same problem. I have 5-6 TidModbusClient connecting to the same TisModbusServer, and the clients randomly hang at reading data from the server. Restarting the client solves the problem.

baur, 2016-07-25:
I'm using the last version (1.6.5) with delphi XE8 + Indy 10, not works!!!

closed 2017-04-05 by plpolak