coassoftwaresystems/delphi-modbus #5: Client: various limits are too short

peter-gsellmann, 2016-04-22:
The client code cuts the request silently without generating exception or error code.
Aside from this, the limits are too small. See attached patch

[0001-various-limits-are-too-short.patch.txt](https://github.com/coassoftwaresystems/delphi-modbus/files/231415/0001-various-limits-are-too-short.patch.txt)

peter-gsellmann, 2016-04-27:
The Server rejects Bit requests with valid length.
The Server returns the wrong length in the answer, so the client detects length inconsistency between request and response.

[0001-Error-on-max-length-and-on-returned-length.patch.txt](https://github.com/coassoftwaresystems/delphi-modbus/files/238587/0001-Error-on-max-length-and-on-returned-length.patch.txt)

plpolak, 2016-05-02:
Unfortunately I cannot test your changes at the moment, so I have accepted and merged them as is.

closed 2016-05-02 by plpolak