coassoftwaresystems/delphi-modbus #11: Install package with RAD Studio 10.1 Berlin (starter edition -> only C++builder , not Delphi)

ModbusOli, 2017-03-07:
Hi,
I have a problem for installing the Modbus package under a RAD Studio 10.1 Berlin (Starter edition -> only  C++Builder 10.1 Berlin , not Delphi): I cannot do the installation as I don't have the Delphi Berlin.
The RAD ide complains with the "Delphi.Personality".
I try to use dcc32 comand line but I am not at all an expert and I fail.
Any chance to have somebody to compile the designtime and the runtime in "bpl" packages for C++Builder  Berlin 10.1? I will appreciate a lot.
Thanks
Best Regards
Olivier

plpolak, 2017-04-05:
I have no experience on C++ builder, so when anyone can provide the correct packages, I can include them.

closed 2017-04-05 by plpolak

Daijobou, 2020-01-22:
I have the same issue with Rad Studio 2009, when you start only C++Builder. This component can only add with Delphi. :(