coassoftwaresystems/delphi-modbus #12: Delphi 10.2 Tokyo

dombleg, 2017-03-27:
Apologies, this is probably the wrong place for this... 
To compile for 10.2 Tokyo, need to add a {$IFDEF VER320} in the ModBusCompiler.inc file.  
I just copied the VER310 section. 
Compiles and installs fine, and on very quick testing works as well.

plpolak, 2017-03-27:
Support for Delphi 10.2 Tokyo is in the works, but to test it I am still waiting for my license from Embarcadero.

closed 2017-03-27 by plpolak

dombleg, 2017-03-27:
Brill. Sorry, new to github.  Thanks for the instant response!   Hopefully you don't anticipate any changes other than to the .inc file?  On the limited testing of the subset of functionality that my program uses, all perfect.  Thankyou!

plpolak, 2017-03-27:
Only changes will be the .inc file. 
In addition new packages have been added for Tokyo