coassoftwaresystems/delphi-modbus #26: Registering Components...

Nuzzi, 2018-04-18:
Great tools, folks.  Thank you!  I am having a problem getting the components to show up in the Tool Palette.  I load the DPK, compile both projects, install and it says it installs fine.  If I check the packages under Options it shows up fine and lists both components, but neither component shows up in the Tool Palette.  Any suggestions?

Nuzzi, 2018-04-18:
I think I just answered my own question. Interesting.  if I have a Delphi project, it shows up fine.  I will create a C++ project and register and see how that goes.

Daijobou, 2020-01-22:
Same here. Successful compiled in Rad Studio 2009 Professional Delphi+C++, but only in Delphi its in tool-palette. In C++Builder its not appear, but old projects have it in forms.

How I get the components in tool-palette for C++Builder?

closed 2024-10-30 by plpolak