coassoftwaresystems/delphi-modbus #33: ReadCoils

lotuschang, 2018-10-18:
I use ReadCoils to get coil message.
coil was 1,3,5,7 to be selected,but array feedback was 2,4,6,8
Please refer my code,has something wrong?
 Var
  iAmount: Integer;
  i: Integer;
  sLine: String;
  myary :array[0..4096] of boolean;
begin
  iAmount := StrToInt(edtReadAmount.Text);
  if (iAmount > 0) then
  begin
    mctPLC.Host := edtIPAddress.Text;
    if  mctPLC.ReadCoils(iAmount,iAmount,myary) then begin
       sLine := 'Register coil(s) read:';
       for i := 0 to (iAmount - 1) do begin
          sLine := sline+ #13#10'     ' +
                 IntToStr(i) +':'+
                 BoolToStr(myary[i],True);
          showmessage(sline);
       end;

JensMertelmeyer, 2018-10-19:
By default, the `BaseRegister` of `TIdModBusServer` is set to `1`. What if you set it to zero?

DeveloppeurPascal, 2020-01-13:
The error is the first parameter of mctPLC.ReadCoils(iAmount,iAmount,myary)