coassoftwaresystems/delphi-modbus #53: The register number to be read is inconsistent with the PLC register number actually read

yangyu-cn-cn, 2022-05-01:
I use Panasonic PLC. When reading the value of a holding register with demo program, it is inconsistent with the actual value. For example, I read the value of register number 3002 and actually returned the value of register number 3001.、

Change the statement at the ‘problem‘ tag to ‘regnumber: = aregnumber;‘ Post problem solving：

function TIdModBusClient.SendCommand(const AModBusFunction: TModBusFunction; const ARegNumber: Word; const ABlockLength: Word; var Data: array of Word): Boolean;
var
  SendBuffer: TModBusRequestBuffer;
  ReceiveBuffer: TModBusResponseBuffer;
  BlockLength: Word;
  RegNumber: Word;
  dtTimeOut: TDateTime;
{$IFDEF DMB_INDY10}
  Buffer: TIdBytes;
  RecBuffer: TIdBytes;
  iSize: Integer;
{$ENDIF}
begin
{$IFDEF DMB_INDY10}
  CheckForGracefulDisconnect(True);
{$ELSE}
  CheckForDisconnect(True, True);
{$ENDIF}
  SendBuffer.Header.TransactionID := GetNewTransactionID;
  SendBuffer.Header.ProtocolID := MB_PROTOCOL;
{ Initialise data related variables }
  RegNumber := ARegNumber - FBaseRegister;      //--------problem--------// 
{ Perform function code specific operations }

JensMertelmeyer, 2022-05-01:
This is not a problem, this is as designed. 

You can adjust the `BaseRegister` property and set it to `0`. The default value is `1`.

yangyu-cn-cn, 2022-05-03:
Thank you for your answer. I was negligent.