coassoftwaresystems/delphi-modbus #57: Fix issues on poor network and poor Modbus controllers

AkLim94, 2022-11-12:
Dear community,

I experienced some issues when using this excellent component.
The issues appended especially on poor network quality and probably poor controllers (with strange behaviors).
I made some changes in the code to fix these issues and would like to share with you, but unfortunately how to contribute....
If anyone know how to do, it's welcome

The changes made:
- Check MBAP header with the length received
- Added more events (to be able to display Modbus frames)
-  Clear input buffer before to send command (to be sure what we receive is for the command we've just sent and not for the previous command, yes some controllers response several times the same)

Regards

plpolak, 2024-10-28:
If you want to have your changes merged, please attach the changes to this ticket, so I can review them for merge.

AkLim94, 2024-10-28:
[IdModbusClient.zip](https://github.com/user-attachments/files/17546288/IdModbusClient.zip)
Hi, I attached my customized unit.
Please search for // Lim
I'm so sorry some comments are in French :)

Best regards

AkLim94, 2025-11-21:
[IdModbusClient.zip](https://github.com/user-attachments/files/23678734/IdModbusClient.zip)

Hi Pipolak,

I saw you're working on the v2. 
As I had a lot of issues like memory corruption, tcp port stuck during disconnection, etc.... with the v1.7.0, 
I made a lot of changes to fix the issues I'm facing.
Also the component is threadsafe now.
I'm glad to share with you what I've done.

Best regards

plpolak, 2025-11-21:
Please check the latest v2 preview, as I think your fixes are already included. If anything is missing, please let me know

AkLim94, 2025-11-22:
Hi, I checked the V2 unit and compared with my new unit version (based on the v1.7.0) provided yesterday, the protections to avoid memory corruption, threadsafe, news methods many compared with the unit I provided on the [Oct 28, 2024]
Please could you check the new one?

An example of test preventing memory corruption
[image]

An example to be threadsafe
[image]

the for loop : " for i := 0 to Length(Buffer) do " is replaced by "for i := 0 to Length(Buffer) - 1 do " <== I need your confirmation
[image]

Some new methods are added

[image]

[image]