coassoftwaresystems/delphi-modbus #65: Can support Modbus RTU over TCP?

seabig6, 2024-06-22:
Can we add support for RTU over TCP, which means adding CRC segments at the end of the data?Can we add support for RTU over TCP, which means adding CRC segments at the end of the data?

seabig6, 2024-07-23:
[source.zip](https://github.com/user-attachments/files/16352362/source.zip)

I try to wirte, could you merge to lib?

plpolak, 2024-08-29:
Looks like your implementation is a copy with addition of the new functionality. Please change it to avoid code duplication: add a new property to enable RTU over TCP, and implement the additional logic in the existing components.

Please also do not touch the formatting of the existing code. This causes a lot of irrelevant updates in git, making it difficult to spot the actual changes.

seabig6, 2024-08-29:
ok