coassoftwaresystems/delphi-modbus #67: License information needs to be added

debiesur, 2024-08-23:
Please add a license to your project so that one can consider using it. Without license it is impossible to use it as part of a commercial product.
Preferably a permissive one like MIT.

JensMertelmeyer, 2024-08-29:
I think the intention is that Github "knows" what license is applied, so you can immediately see on the front page, or filter repos by license, ...

[image]

https://docs.github.com/en/communities/setting-up-your-project-for-healthy-contributions/adding-a-license-to-a-repository

plpolak, 2024-08-29:
I moved the license file to the root of the repo, and now it is recognized by GitHub

closed 2024-08-29 by plpolak